lifelong-robotic-vision/OpenLORIS-Scene #3: Fault rosbag play hz

oym1994, 2020-01-09:
The real D435i color camera sample frequency is about 32hz, while the rosbag play frequency of /d400/color/image_raw is only about 8hz.  It shows that the consuming time of an algorithm can be more than 150ms if using roabag data, which will  be likely to be not  real time  if using the camare in real situation. Have u taken this problem into consideration? Thank u!!!

oym1994, 2020-01-09:
or there is  another command to play the rosbag data in real time mode instead of  "rosbag play xxx --clock".

cedrusx, 2020-01-09:
@oym1994 Thank you for trying out with OpenLORIS.
I could not reproduce your issue - when I play bags with the following commands, the average rate is around 30.
```
rosbag play openloris/home-1/bags/home1-1.bag
rostopic hz /d400/color/image_raw
```
Could you provide more details? Including which bag you are playing, what other workloads are there on your PC (e.g. if a SLAM algorithm is running), OS version, CPU and disk information (if you are using a portable hard drive then the limited bandwidth might be an issue), etc.

oym1994, 2020-02-05:
Thank u !   

rosbag play  home1-1-all.bag
rostopic hz /d400/color/image_raw

But the avarege frequency is about 10hz.

No other workloads . Here is my PC information:
Dell G7-7588 ； intel i7-8750H @2.2GHz；NVIDIA GTX 1060i;  ubuntu16.04(Not in a virture machine);
256G SSD + 1T mechanical hard disk  

I can't download the home1-1.bag now. Maybe home1-1-all.bag is not the same as home1-1.bag?

Thank u and take good care of  yourself in such a bad situation.

cedrusx, 2020-02-05:
@oym1994 Thanks for the information and for the caring!

Could you identify, e.g. by examining the published images in rviz, whether the low FPS is caused by frame drops or message delays? If latter, maybe could try out with copying the bag to the SSD.

Below is the bag info of home1-1.bag. The image topics in home1-1-all.bag should be the same.
```
$ rosbag info home1-1.bag
path:        home1-1.bag
version:     2.0
duration:    2:32s (152s)
start:       Jun 08 2019 21:20:02.51 (1560000002.51)
end:         Jun 08 2019 21:22:35.49 (1560000155.49)
size:        18.0 GB
messages:    175238
compression: none [18377/18377 chunks]
types:       nav_msgs/Odometry         [cd5e73d190d741a2f92e81eda573aca7]
             realsense2_camera/IMUInfo [b8a77fbb6e4146ae79c8a943413e4f62]
             sensor_msgs/CameraInfo    [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image         [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu           [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan     [90c7ef2dc6895d81024acba2ac42f369]
             tf2_msgs/TFMessage        [94810edda583a504dfda3829e70d7eec]
topics:      /d400/accel/imu_info                           1 msg     : realsense2_camera/IMUInfo
             /d400/accel/sample                         38998 msgs    : sensor_msgs/Imu
             /d400/aligned_depth_to_color/camera_info       1 msg     : sensor_msgs/CameraInfo
             /d400/aligned_depth_to_color/image_raw      4587 msgs    : sensor_msgs/Image
             /d400/color/camera_info                        1 msg     : sensor_msgs/CameraInfo
             /d400/color/image_raw                       4587 msgs    : sensor_msgs/Image
             /d400/depth/camera_info                        1 msg     : sensor_msgs/CameraInfo
             /d400/depth/image_raw                       4587 msgs    : sensor_msgs/Image
             /d400/gyro/imu_info                            1 msg     : realsense2_camera/IMUInfo
             /d400/gyro/sample                          60989 msgs    : sensor_msgs/Imu
             /gt                                         3077 msgs    : tf2_msgs/TFMessage
             /odom                                       3056 msgs    : nav_msgs/Odometry
             /scan                                       6123 msgs    : sensor_msgs/LaserScan
             /t265/accel/imu_info                           1 msg     : realsense2_camera/IMUInfo
             /t265/accel/sample                          9546 msgs    : sensor_msgs/Imu
             /t265/fisheye1/camera_info                     1 msg     : sensor_msgs/CameraInfo
             /t265/fisheye1/image_raw                    4590 msgs    : sensor_msgs/Image
             /t265/fisheye2/camera_info                     1 msg     : sensor_msgs/CameraInfo
             /t265/fisheye2/image_raw                    4590 msgs    : sensor_msgs/Image
             /t265/gyro/imu_info                            1 msg     : realsense2_camera/IMUInfo
             /t265/gyro/sample                          30498 msgs    : sensor_msgs/Imu
             /tf_static                                     1 msg     : tf2_msgs/TFMessage
```

oym1994, 2020-02-23:
Thank you. I have found the problem.  I used the compressed version dataset.  Only 4.9G. And after being decompressed,  it's OK.

`path:         home-1-1-all.bag
version:      2.0
duration:     2:32s (152s)
start:        Jun 08 2019 21:20:02.50 (1560000002.50)
end:          Jun 08 2019 21:22:35.50 (1560000155.50)
size:         4.9 GB
messages:     166038
compression:  bz2 [18352/18352 chunks; 27.08%]
uncompressed: 18.0 GB @ 120.6 MB/s
compressed:    4.9 GB @  32.7 MB/s (27.08%)
types:        nav_msgs/Odometry         [cd5e73d190d741a2f92e81eda573aca7]
              realsense2_camera/IMUInfo [b8a77fbb6e4146ae79c8a943413e4f62]
              sensor_msgs/CameraInfo    [c9a58c1b0b154e0e6da7578cb991d214]
              sensor_msgs/Image         [060021388200f6f0f447d0fcd9c64743]
              sensor_msgs/Imu           [6a62c6daae103f4ff57a132d6f95cec2]
              tf2_msgs/TFMessage        [94810edda583a504dfda3829e70d7eec]
topics:       /d400/accel/imu_info                           1 msg     : realsense2_camera/IMUInfo
              /d400/accel/sample                         38998 msgs    : sensor_msgs/Imu          
              /d400/aligned_depth_to_color/camera_info       1 msg     : sensor_msgs/CameraInfo   
              /d400/aligned_depth_to_color/image_raw      4587 msgs    : sensor_msgs/Image        
              /d400/color/camera_info                        1 msg     : sensor_msgs/CameraInfo   
              /d400/color/image_raw                       4587 msgs    : sensor_msgs/Image        
              /d400/depth/camera_info                        1 msg     : sensor_msgs/CameraInfo   
              /d400/depth/image_raw                       4587 msgs    : sensor_msgs/Image        
              /d400/gyro/imu_info                            1 msg     : realsense2_camera/IMUInfo
              /d400/gyro/sample                          60989 msgs    : sensor_msgs/Imu          
              /odom                                       3056 msgs    : nav_msgs/Odometry        
              /t265/accel/imu_info                           1 msg     : realsense2_camera/IMUInfo
              /t265/accel/sample                          9546 msgs    : sensor_msgs/Imu          
              /t265/fisheye1/camera_info                     1 msg     : sensor_msgs/CameraInfo   
              /t265/fisheye1/image_raw                    4590 msgs    : sensor_msgs/Image        
              /t265/fisheye2/camera_info                     1 msg     : sensor_msgs/CameraInfo   
              /t265/fisheye2/image_raw                    4590 msgs    : sensor_msgs/Image        
              /t265/gyro/imu_info                            1 msg     : realsense2_camera/IMUInfo
              /t265/gyro/sample                          30498 msgs    : sensor_msgs/Imu          
              /tf_static                                     1 msg     : tf2_msgs/TFMessage`

oym1994, 2020-02-23:
Thank you. I have found the problem. I used the compressed version dataset. Only 4.9G. And after being decompressed, it's OK.

rosbag info home1-1-all.bag

`path:         home-1-1-all.bag
version:      2.0
duration:     2:32s (152s)
start:        Jun 08 2019 21:20:02.50 (1560000002.50)
end:          Jun 08 2019 21:22:35.50 (1560000155.50)
size:         4.9 GB
messages:     166038
compression:  bz2 [18352/18352 chunks; 27.08%]
uncompressed: 18.0 GB @ 120.6 MB/s
compressed:    4.9 GB @  32.7 MB/s (27.08%)
`

oym1994, 2020-02-23:
Thank you. I have found the problem. I used the compressed version dataset. Only 4.9G. And after being decompressed, it's OK.

$rosbag info home1-1-all.bag

path:    home-1-1-all.bag
version:      2.0
duration:     2:32s (152s)
start:        Jun 08 2019 21:20:02.50 (1560000002.50)
end:          Jun 08 2019 21:22:35.50 (1560000155.50)
size:         4.9 GB
messages:     166038
compression:  bz2 [18352/18352 chunks; 27.08%]
uncompressed: 18.0 GB @ 120.6 MB/s
compressed:    4.9 GB @  32.7 MB/s (27.08%)

cedrusx, 2020-02-24:
@oym1994 Cool! Thank you for the update. I'll close the issue and add a note onto the data index.

closed 2020-02-24 by cedrusx