lifelong-robotic-vision/OpenLORIS-Scene #4: Can you share how to download the dataset?

zxy-HIT, 2020-01-14:
I tried this website, but couldn't open it. https://lifelong-robotic-vision.github.io/dataset/scene.html
Can you help me find out the right way to download them?

cedrusx, 2020-01-14:
Hi @zxy-HIT , please fill [this form](http://openloris.sv.mikecrm.com/Nv8LzXF) to download the data.
Since you can access github, I guess you may be able to open [the source code page of the website](https://github.com/lifelong-robotic-vision/lifelong-robotic-vision.github.io/blob/master/dataset/scene.md) where you can view all the contents.

zxy-HIT, 2020-01-14:
Hi, Thank you so much for your help, but I just couldn't open the website of "this form", is this because I am in Singapore now?  Can you help me?

cedrusx, 2020-01-14:
Hi @zxy-HIT, not sure about the cause but we did have registrants from Singapore. Could you ask a friend to try out with the link and help fill the form? Or you may email to me the information required in the following screenshot and I can fill it for you.

[image]

zxy-HIT, 2020-01-14:
> Hi @zxy-HIT , please fill [this form](http://openloris.sv.mikecrm.com/Nv8LzXF) to download the data.
> Since you can access github, I guess you may be able to open [the source code page of the [...]

Now, I get that, it is because I am not in China. When I use vpn, I can open it. Thank you so much!

cedrusx, 2020-01-14:
> Now, I get that, it is because I am not in China. When I use vpn, I can open it. Thank you so much!

No problem. Thank you for your interest in OpenLORIS!

closed 2020-01-14 by cedrusx