lifelong-robotic-vision/OpenLORIS-Scene #5: data style of file  in  odom.txt ???

lishuwei0424, 2020-04-05:
Hi, thank you  for open  datasets, I download office-1-2,  data   style  in  odom.txt as follows:
__________________________
 #Time pose.position.x y z pose.orientation.x y z w twist.linear.x y z twist.angular.x y z
1560000122.95948553 19.2181491852 22.0353050232 0.0 0.0 0.0 -0.939260749489573 0.343204377111198 -0.08578508240731442 -0.07235155355804822 0.0 0.0 0.0 0.209613412619
__________________________
start point is  no zero , I am feel uncertain，Is it convenient to answer my  question ?

cedrusx, 2020-04-05:
Hi @lishuwei0424 , thanks for the question. The data sequences in OpenLORIS were truncated from longer trajectory recordings, so it is common that the odom data do not start from origin. Please use the relative positions within a sequence.

lishuwei0424, 2020-04-06:
Hi@cedrusx , I konw it .  thank you  for you reply .

closed 2020-04-06 by lishuwei0424