lifelong-robotic-vision/OpenLORIS-Scene #6: Depth factor for aligned_depth  images

ankuPRK, 2020-05-01:
Hi when I read the depth images for various scenes (office, home, etc), using:
imdepth = cv2.imread(path, cv2.IMREAD_UNCHANGED)

I get int64 values from 0 to around 5700. What is the depth factor i.e. how do I convert these pixel values to actual depth in meters?

Is the depth factor 1000.0?

cedrusx, 2020-05-02:
Hi @ankuPRK, yes, the depth factor is 1000.0. You may simply convert the int64 values into float and multiply them by 0.001 to get depth in meters. The depth value of 0 means invalid.

Refer to this code which converts the depth values (`in`) into meters (`out`) and down-sample it by `scale_ratio`. It also filters out depth values larger than `depth_max_range` set by the user. The max possible range from RealSense D400 is 10 meters.

```c++
void Frame::convertIntoMetersAndScale(
        float *out, const ushort *in,
        unsigned int width, unsigned int height,
        unsigned int scale_ratio, float depth_max_range)
{
    static const float scale_to_meters = 0.001;
    const unsigned int out_w = width / scale_ratio;
    const unsigned int out_h = height / scale_ratio;
    const unsigned int width_multiply_ratio = width * scale_ratio;
    unsigned int y;
    for (y = 0; y < out_h; y++) {
        const unsigned int in_offset = width_multiply_ratio * y;
        const unsigned int out_offset = out_w * y;
        for (unsigned int x = 0; x < out_w; x++) {
            float dist = (float)in[x * scale_ratio + in_offset] * scale_to_meters;
            out[x + out_offset] = dist < depth_max_range ? dist : 0;
        }
    }
}
```

ankuPRK, 2020-05-02:
Thanks! @cedrusx

closed 2020-05-02 by ankuPRK