lifelong-robotic-vision/OpenLORIS-Scene #7: Is it possible to sync the groundtruth.txt with the color images?

amishra791, 2020-05-14:
I noticed that the color images (taken by D435i device) correspond to timestamps not found in the groundtruth.txt. Is there any way to reconcile the two together?

cedrusx, 2020-05-14:
Hi @amishra791 , the groundtruth values are from mocap (for office1) or laser (for others) measurements. They generally have different timestamps from the D435i or T265 images. Would it make sense for your application if simply finding the groundtruth with nearest timestamp, or interpolating between two nearest values?

amishra791, 2020-05-14:
Thank you for your response. Are the timestamp measurements corresponding to the D435i images accurate (i.e. 60.png was really taken at time 60)?

cedrusx, 2020-05-15:
> Thank you for your response @cedrusx . Are the timestamp measurements corresponding to the D435i images accurate (i.e. 60.png was really taken at time 60)?

Yes, I suppose so. The data from RealSense cameras are hardware synchronized and stamped. The sync error between RealSense data and groundtruth should be within several milliseconds per our estimation. But the groundtruth values from laser SLAM may not be as accurate as one may expect. Please refer to the paper for more explanation.

amishra791, 2020-05-15:
Will take a look at the paper. Thank you for your help!

closed 2020-05-15 by amishra791