lifelong-robotic-vision/OpenLORIS-Scene #10: What's the meaning of 'aligned_depth', and how to Synchronize the depth and color images?

KaninchenM, 2020-06-22:
Hi. Thanks for your data. I've downloaded the Tum Format Package. Before using it, I've got some questions. 
1. What's the meaning of 'aligned_depth'? What's the difference between the data in folder `aligned_depth` and `depth`? I found they had the same count of images.
2. How to synchronize the depth and color images? I think as the data is Tum Format like, maybe I should use the `Tum associating tool`[here](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools). 
3. If my idea in Q2 is right, which depth folder should I use to associate with RGB images in folder `color`? As the information in `depth.txt`, maybe I should use images in `depth`. If so, what information the images in `aligned_depth` aligned?
The information in `depth.txt` likes:
`1560000083.949189 depth/1560000083.949189.png
1560000083.982551 depth/1560000083.982551.png`

Waiting for your reply. Thank you so much.

cedrusx, 2020-06-25:
Hi @KaninchenM , the aligned depth folder includes post-processed depth images which has been re-projected into the RGB camera frame so that there is a per-pixel correspondence (i.e., pixel(i,j) in aligned_depth/t.png corresponds to pixel(i,j) in color/t'.png). So if you need RGBD then most likely you should use the aligned depth than the (raw) depth.

The color images and depth images are synchronized. For each color image, there is one corresponding depth/aligned_depth image which has approximately the same time stamp (diff less than 1ms). You can use the TUM association tool if you like.

KaninchenM, 2020-06-29:
> Hi @KaninchenM , the aligned depth folder includes post-processed depth images which has been re-projected into the RGB camera frame so that there is a per-pixel correspondence (i.e., pixel(i,j) in [...]

Thank you. I've used the TUM association tool to associate aligned depth images and depth images.

closed 2020-06-30 by cedrusx