lifelong-robotic-vision/OpenLORIS-Scene #11: How to get the RGBD camera parameters?

KaninchenM, 2020-07-13:
Thanks for contrbuting the life-long slam data. I'm going to use office1-7 to run orbslam2, which need camera parameters such as 
[image]

My questions are:
1. How can I get RGBD camera inner parameters?
    I get following from `sensors.yaml`. But which one should I use, the `d400_color_optical_frame` or `d400_depth_optical_frame`?
`Camera.fx: 6.1145098876953125e+02`
`Camera.fy: 4.3320397949218750e+02`
`Camera.cx: 6.1148571777343750e+02`
`Camera.cy: 2.4947302246093750e+02`
   So do the Distortion parameters.
2. How can I get the `Camera.bf` and `ThDepth`?
   I get them from here: [Begin ORBSlam2](https://www.iambigboss.top/post/22545_1_1.html). In this url, I find the parameters of D435i. So I set:
   `Camera.bf: 30.6710083`
   `ThDepth: 50.0`
   But the inner parameters and distortion parameters are different from which I find in Open-Scene Office 's `sensors.yaml`. Why? Which one should I use?
Waiting for your reply. Thank you.

cedrusx, 2020-07-13:
Hi @KaninchenM , for running ORB_SLAM2 you should use the aligned_depth, which has a per-pixel alignment with the corresponding color image. So, please use the intrinsics of `d400_color_optical_frame`.

`Camera.bf` is baseline * fx. The baseline of RealSense D435i is 0.05 (5 cm), so `bf = 0.05 * 6.1145098876953125e+02 = 30.57`.
`ThDepth` is an empirical param, which we do not have any particular suggestion. I set 60 in my settings but it is just an arbitary choice.
`DepthMapFactor` should be 1000.

KaninchenM, 2020-07-14:
> Hi @KaninchenM , for running ORB_SLAM2 you should use the aligned_depth, which has a per-pixel alignment with the corresponding color image. So, please use the intrinsics of [...]

Thank you very much! It really helps me.

closed 2020-07-15 by cedrusx

dinarkino, 2020-07-23:
By the way, @KaninchenM you should change Camera.fy and Camera.cx. I think, they posted in row-wise order in the config file

KaninchenM, 2020-07-27:
> By the way, @KaninchenM you should swap values of Camera.fy and Camera.cx. I think, values posted in row-wise order in the config file

As you say, the `data` matrix in `sensor.yaml`
[image]
gave as the order `fx,cx,fy,cy`? No wonder my experiment results looks not good. I'll try it today. Thank you so much!

dinarkino, 2020-07-27:
I suppose yes, they are in such an order and usually, fx and fy have similar values 
[image]

KaninchenM, 2020-07-27:
> I suppose yes, they are in such an order and usually, fx and fy have similar values
> [image]
> [opencv](https://vovkos.github.io/doxyrest-showcase/opencv/sphinx_rtd_theme/page_tutorial_camera_calibration.html)

I've fixed my parameters of camera, and got the results rmse=0.114 in sequence office7. I think it's similar to the result in the paper. Thank you so much! I spent much time to find the cause of my difference.

KaninchenM, 2020-07-27:
> Hi @KaninchenM , for running ORB_SLAM2 you should use the aligned_depth, which has a per-pixel alignment with the corresponding color image. So, please use the intrinsics of [...]

Sorry to bother you again. I am using [evo](https://github.com/MichaelGrupp/evo/wiki/Metrics). Did your team also use it as the tool to evaluate the results? If so, which `align` model do you use in RGBD sequence? `-a`(SE(3) Umeyama alignment (rotation, translation)) or `-as`( Sim(3) Umeyama alignment (rotation, translation, scale))?