lifelong-robotic-vision/OpenLORIS-Scene #13: coordinate transformation

Grandzxw, 2020-07-21:
Do the coordinates of the wheel odometer represent the base_link coordinates?

cedrusx, 2020-07-21:
Yes they do

Grandzxw, 2020-07-22:
One more question: Can the angular and linear velocity integrals of a wheel directly get the wheel's pose? or the IMU data needs to be added？

cedrusx, 2020-07-22:
> One more question: Can the angular and linear velocity integrals of a wheel directly get the wheel's pose? or the IMU data needs to be added？

`base_link` poses have already been given in the odometry data (in the `base_odom` frame). No need to do any integration if you want a trajectory from wheel odemetry.

Grandzxw, 2020-07-22:
Is the IMU data added to the wheel pose? If you're going to get a direct integral without IMU data, can you get a  pose?

cedrusx, 2020-07-22:
> Is the IMU data added to the wheel pose? If you're going to get a direct integral without IMU data, can you get a pose?

For home/cafe/office/corridor, the odometry data were fused from data from wheel encoders and an IMU on the base. For market, the odometry data were directly from wheel encoders.

cedrusx, 2020-10-22:
closing as no update for months - feel free to re-open

closed 2020-10-22 by cedrusx