lifelong-robotic-vision/OpenLORIS-Scene #14: Raw wheel odometry

dinarkino, 2020-07-27:
Is there a way to get data such as ticks from wheel encoders?

cedrusx, 2020-07-29:
Hi @dinarkino , we did record the raw ticks data, but unfortunately they had been excluded from the very early stage of data processing. It would require quite a lot of efforts to get them back from the raw database. We could do that if there is a strong need, but maybe not recently.

The given odom data in the dataset were fused from ticks and IMU measurements, so it would not be possible to recover the original ticks from odom. But if you just want data in ticks format then a conversion may work.

dinarkino, 2020-07-29:
Thank you for the reply! There is no strong need, I just wanted to run an algorithm (DRE-SLAM) that uses raw ticks.

closed 2020-07-29 by dinarkino