lifelong-robotic-vision/OpenLORIS-Scene #15: Parameters for VINS-MONO

arindamsaha81, 2020-08-07:
Hello,
Thanks for a nice dataset. I was trying to test the performance of VINS-MONO on your dataset. I am bit confused to use the values for acc_n, acc_w, gyr_n, gyr_w these are nothing but measurement noise standard deviation and bias random work noise standard deviation of accelerometer and gyro. Can you please help me to get these values for your D435i?

Thanks in advance.

arindamsaha81, 2020-08-24:
Hi **@cedrusx**, Thanks for the info. I have already checked the sensor.yaml and little confused about the intrinsic 1x12 matrix as because I am very new to work with IMU. Can you please explain about the values that the matrix contains or any link that explain the properties of the matrix. For example which values to pick up from the 1x12 matrix for acc_n, acc_w and similar for gyroscope

cedrusx, 2020-09-13:
Hi @arindamsaha81 , you may find an explanation of the IMU parameters in the FAQ section of the [OpenLORIS-Scene webpage](https://lifelong-robotic-vision.github.io/dataset/scene), under the question of "What are the variances in IMU intrinsics". I hope it helps.

arindamsaha81, 2020-09-25:
Hi **@cedrusx** thanks for the info. The information on the FAQ tells about data units but does not explain about the 1x12 matrix. I am interested in to know the imu_intrinsic matrix description. The paper tells the IMU intrinsics are from factory calibration. So I assume, the 1x12 matrix is populated from factory calibration. But how and where did you get those values?

arindamsaha81, 2020-09-27:
Hi **@cedrusx**, The paper "Are We Ready for Service Robots? The OpenLORIS-Scene Datasets for Lifelong SLAM" tells that VINS-MONO is executed on every sequences and I am trying to run VINS-MONO. Therefore, can you please provide me the values that you used for acc_n, acc_w, gyr_n, gyr_w for D435i and T265. This would be good enough for me to run VINS-MONO successfully.

cedrusx, 2020-10-22:
> Hi **@cedrusx** thanks for the info. The information on the FAQ tells about data units but does not explain about the 1x12 matrix. I am interested in to know the imu_intrinsic matrix description. [...]

Hi @arindamsaha81 sorry for the delayed response. The 1x12 matrix should actually be a 3x4 matrix with the left 3x3 block as a rotation matrix to rectify the axes, and the right 3x1 block as zero offsets. So you may left multiply the 3x4 matrix to the homogeneous data vector to get rectified data. The intrinsic values were dumped from the device's NVRAM with `rs-enumerate-devices -c`. For T265 they were from factory calibration, while for D435i they were from our manual calibration with the [official calibration tool](https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool).

cedrusx, 2020-10-22:
> Hi **@cedrusx**, The paper "Are We Ready for Service Robots? The OpenLORIS-Scene Datasets for Lifelong SLAM" tells that VINS-MONO is executed on every sequences and I am trying to run VINS-MONO. [...]

Just checked with my co-author who had run VINS-Mono, turned out that the configuration files have been lost. :(  I'm very sorry. As I remember it, we did not take much time tuning the parameters. Might be using the [sample configurations from VINS-Mono repo](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/tree/master/config/realsense) or partially from the variances given by RealSense, or some values in between. Our results were not quite good anyway, so perhaps there can be better configruations.

JEREMY824, 2022-12-24:
> 嗨**@cedrusx**，论文“我们为服务机器人做好准备了吗？OpenLORIS-Scene Datasets for Lifelong SLAM“告诉VINS-MONO在每个序列上执行，我正在尝试运行VINS-MONO。因此，您能否提供您用于 D435i 和 T265i 的acc_n、acc_w、gyr_n gyr_w的值。这足以让我成功运行VINS-MONO

Did you run VINS successfully? I also have the same question, can you easily answer it

lucienyoung, 2023-06-25:
Hi, as we know, the 1x12 matrix stores 3x3 rotation matrix and 1x3 translation vector respectively.  I'd like to know whether we can simply left multiply the rotation matrix to the homogeneous data vector, i.e. the noise variance vector of gyroscope, and then add the 1x3 translation vector to rectify the data from x, y and z axis. Finally, by taking an average we can get the gyr_n. Other intrinsic values can be obtained in the same way. I don't know whether or not this method would be correct, can anyone comment on this?

lucienyoung, 2023-06-25:
> > 嗨**@cedrusx**，论文“我们为服务机器人做好准备了吗？OpenLORIS-Scene Datasets for Lifelong SLAM“告诉VINS-MONO在每个序列上执行，我正在尝试运行VINS-MONO。因此，您能否提供您用于 D435i 和 T265i 的acc_n、acc_w、gyr_n gyr_w的值。这足以让我成功运行VINS-MONO
> 
> Did you run VINS successfully? I also have the same question, can you easily answer it

Hello, did you address this issue? Please see from my comment.

lucienyoung, 2023-06-25:
Hi, as we know, the 1x12 matrix stores 3x3 rotation matrix and 3x1 translation vector respectively. I'd like to know whether we can simply left multiply the rotation matrix to the homogeneous data vector, i.e. the noise variance vector of gyroscope, and then add the 3x1 translation vector to rectify the data from x, y and z axis. Finally, by taking an average we can get the gyr_n. Other intrinsic values can be obtained in the same way. I don't know whether or not this method would be correct, can anyone comment on this?