lifelong-robotic-vision/OpenLORIS-Scene #18: Combining accelerometer and gyroscope

arindamsaha81, 2020-08-24:
Hello,

I m trying to run VINS-RGBD with your rosbag. The slam code expect accelerometer and gyroscope to be in a single topic. I get /camera/imu topic by linear_interpolation of /camera/accel/sample and /camera/gyro/sample which is provided by the realsense-ros. But your rosbag contains accelerometer and gyroscope on separate topics. Do you have any script or any application that combine them with linear_interpolation into a single topic?

Thanks in advance.

cedrusx, 2020-09-13:
Hi, please check [this script](https://gist.github.com/cedrusx/da24308e1824a6fabb0aa9f6ed7042b0).

Usage: ./unite_imu_topics.py office1-1.bag office1-2.bag ...

arindamsaha81, 2020-09-25:
Hi **@cedrusx** Thanks a lot.

closed 2020-09-26 by cedrusx

504115681, 2021-06-01:
Hello,when I execute "./unite_imu_topics.py corridor1-1.bag" ,it can not stop,and I force the use of "Ctrl+c" in the terminal,it

504115681, 2021-06-01:
Hello,when I execute "./unite_imu_topics.py corridor1-1.bag" ,it can not stop,and I force the use of "Ctrl+c" in the terminal,it shows stop at this step: “if target_t >=pre_t and target_t<=last_t:”.

ScottishOwl, 2026-03-14:
> Hi, please check [this script](https://gist.github.com/cedrusx/da24308e1824a6fabb0aa9f6ed7042b0).
> 
> Usage: ./unite_imu_topics.py office1-1.bag office1-2.bag ...

Hi, I am also trying to combine /camera/accel/sample and /camera/gyro/sample in one topic.
However, this script is using Python2 + ROS1, while my configuration is Python3 + ROS2 Humble. Consequently I cannot directly run the script to unite the imu topics.
Is there a newer version of this script that can run in Python3 + ROS2 environment?