lifelong-robotic-vision/OpenLORIS-Scene #19: Understanding trans_matrix.yaml

ankuPRK, 2020-08-24:
Hey guys could you explain what "base_link" and "marker" mean in trans_matrix.yaml with different sequences. 

Also, for a given parent_frame and child_frame, is the extrinsic matrix T_[parent][child] or reverse i.e. if I have coordinates x_[child] then is the extrinsic matrix given here of the form:
x_[parent] = T_[parent][child]*x_[child]
or vice versa?

cedrusx, 2020-09-13:
Hi @ankuPRK , `base_link` is the center of the two driving wheels. `marker` is the centroid of the several OptiTrack markers on top of the robot. You may check the picture in the middle of [the webpage](https://lifelong-robotic-vision.github.io/dataset/scene) to get an intuitive idea.

It is T_[parent][child], following the ROS tf convention.

cedrusx, 2020-10-22:
closing as no update for weeks - feel free to re-open

closed 2020-10-22 by cedrusx

ankuPRK, 2020-10-22:
Hi @cedrusx sorry for forgetting to close the issue! Yes your response on Sep 12 makes sense. Thank you! :)