lifelong-robotic-vision/OpenLORIS-Scene #21: Lidar measurements for OpenLoris-Scene dataset?

FrederikWarburg, 2020-09-23:
Hi

Thank you for realizing this dataset. 

It seems that the robot is equipped with a LiDAR scanner, however, when I look at the downloaded data, there doesn't seem to be any LiDAR data available. Is it possible to realize the LiDAR measurements?

In the downloaded dataset folder I have the following files:

[image]

cedrusx, 2020-09-24:
Oh we forgot to add the laser data into the package. My apologies for that. Would it be convenient for you to use ROS bags? Meanwhile we will try to re-build the package and update on Google Drives but it may take a few weeks.

FrederikWarburg, 2020-09-24:
I would prefer the packaged version, similar to the TUM format. Thank you.