lifelong-robotic-vision/OpenLORIS-Scene #24: Laser scans missing from market rosbags

khaled-98, 2020-10-22:
Hello,

Thank you so much for providing these datasets!

I was hoping to use the latest available rosbag for the market dataset but it seems to be missing the laser scans. Is there somewhere else where I can get this data from?

Khaled

cedrusx, 2020-12-11:
Hi @khaled-98 , thanks for reporting this issue. It seems that the scan data of market have been omitted since the beginning of our data processing. Fortunately we have recovered them from the original recordings. [Here they are](https://drive.google.com/file/d/1JEAs0r071cRCcs3Dw34wXM-xw1LUGgN5/view?usp=sharing).

The data were from a RS-LIDAR, with a special message format. The [ros_rslidar](https://github.com/RoboSense-LiDAR/ros_rslidar) package will be needed to get point cloud from these messages.

Would you let me know if you'd be able to give it a try? Thanks again and apologize for the inconveniences.

yuhui-xie, 2021-11-01:
> Hi @khaled-98 , thanks for reporting this issue. It seems that the scan data of market have been omitted since the beginning of our data processing. Fortunately we have recovered them from the [...]

Do you have collected 2D laser data in the marker sequence? If so could you please release the data? Thank you very much.

cedrusx, 2021-11-06:
> Do you have collected 2D laser data in the marker sequence? If so could you please release the data? Thank you very much.

No, the robot collecting market data did not have 2D laser scanner.