lifelong-robotic-vision/OpenLORIS-Scene #34: Question about DepthMapFactor

UESTCgr, 2022-11-03:
I set DepthMapFactor (what pixel value in the depth image corresponds to 1m) as 1000.0 and evaluate ORB-SLAM2(official version)  on OpenLoris dataset, but it seems the scale is incorrect. The result is given by your evaluation tool:
[image]

I adjusted DepthMapFactor to 800 and it seems fine:
[image]

The major camera parameters to run ORB_SLAM2 are:
```
Camera.fx: 6.1145098876953125e+02
Camera.fy: 6.1148571777343750e+02
Camera.cx: 4.3320397949218750e+02
Camera.cy: 2.4947302246093750e+02
Camera.width: 848
Camera.height: 480
Camera.imageScale: 0.5
ThDepth: 50.0
Camera.bf: 30.57
DepthMapFactor: 800
```

But the assumed value(800) may not correct, thus my evaluation will be inaccurrate. Could you please publish the correct DepthMapFactor, or the scale error is caused by other reasons. 

Thank you!

cedrusx, 2022-11-11:
@UESTCgr I'm pretty confident that 1000 is correct, and the intrinsics in your config are correct as well. We use OpenLORIS-Scene in many RGBD experiments and have not observed scale issue. Could you check you other settings?
Your ThDepth might be a bit large. I suggest using a smaller value, e.g. 20, because the baseline of RealSense is 5cm only.
What is Camera.imageScale? I don't think the official ORBSLAM2 has such a param.

lichiahui, 2025-01-14:
[image]
@UESTCgr @cedrusx I meet the same problem with ORBSLAM3(rgbd), have you solve it?
```
Camera1.fx: 611.4509887695312
Camera1.fy: 611.4857177734375
Camera1.cx: 433.2039794921875
Camera1.cy: 249.4730224609375

# distortion parameters
Camera1.k1: 0.0
Camera1.k2: 0.0
Camera1.k3: 0.0
Camera1.p1: 0.0
Camera1.p2: 0.0

# Camera resolution
Camera.width: 848
Camera.height: 480

# Camera frames per second 
Camera.fps: 30
#0.05
#Camera.bf: 30.57255

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1
#specifies the maximum depth in meters allowed for a point
System.thFarPoints: 3.0

# Close/Far threshold. Baseline times.
Stereo.ThDepth: 20.0
Stereo.b: 0.0745

# Depth map values factor
RGBD.DepthMapFactor: 1000.0
```